Xilinx/llvm-aie #126: Support for software pipelining in pragmas.

aidansander, 2024-07-17:
I'm compiling a simple kernel using peano. Manually software pipelining the attached kernel (dut_pipelined.cc) yields considerable speedup compared to using pipelining pragmas (dut_pragma.cc). Without manual pipelining, the produced assembly does not pipeline and the kernel runs in ~1800 cycles. With manual pipelining, the kernel runs in ~1000 cycles. The `clang loop min_iteration_count` and `max_iteration_count` pragmas have no effect on the produced assembly. 
[dut_pragma.cc](https://github.com/user-attachments/files/16275144/dut_pragma.txt)
[dut_pipelined.cc](https://github.com/user-attachments/files/16275146/dut_pipelined.txt)

konstantinschwarz, 2024-07-17:
Hi @aidansander, 
could you also provide the `lut_based_ops.h` header to be able to reproduce your results?

aidansander, 2024-07-18:
Sure thing. lut_based_ops.h and lut_based_ops.cpp (which holds the LUT values) are both included from [here](https://github.com/aidansander/mlir-aie/tree/main/aie_runtime_lib/AIE2). I'm running the kernel using llvm-lit on some tests. [The manually pipelined](https://github.com/aidansander/mlir-aie/blob/main/test/unit_tests/aievec_tests/bf16_exp_lut_llvm/dut_scheduling_tests/003_pragma_unroll/clang_rescheduled.mlir) and [loop pragma](https://github.com/aidansander/mlir-aie/blob/main/test/unit_tests/aievec_tests/bf16_exp_lut_llvm/pragma_tests/006_unroll_iteration_count/bf16_exp_lut_llvm.mlir) tests have the steps I used to run and measure the cycle count.

konstantinschwarz, 2024-07-19:
Thanks!
One thing to consider: LoopUnroll runs before MachinePipeliner, i.e. if you completely unroll the loop (through `#pragma unroll`), there is nothing to do for the pipeliner.

There are a few things we are still lacking though to get this software pipelined:
* Enable hardware loops by default, to turn an up-counting loop into a down-counting loop. Currently blocked by the next item.
* Teach the MachinePipeliner to understand our hardware loop construct (we have an open PR for this https://github.com/Xilinx/llvm-aie/pull/125)
* VLDB4X instructions don't carry MachineMemOperands, pessimizing alias analysis results in the MachinePipeliner and rejecting the loop.

With these tweaks, I could get a SWP loop with 23 cycles. It should be possible to further improve on that.
FYI @gbossu @martien-de-jong @andcarminati

konstantinschwarz, 2024-08-05:
Update: 1. & 2. have been resolved.
Last point - MachineMemOperands for VLDB.4x instructions - still needs to be done.