Xilinx/llvm-aie #160: Convolution 2D kernel use case

jsetoain, 2024-08-20:
This use case is a
[vectorized conv2d kernel](https://github.com/user-attachments/files/16674182/conv2d-use-case.zip). It's lowered from MLIR without using any target-specific dialect (like AIEVec), and it presents a couple of interesting challenges.

The first thing that trips the compiler is `%143:_(<32 x s32>) = G_ADD %142:_, %112:_`, which looks like a 1024b vector addition. Those are supported (on accumulator registers), so it should be possible to select for them.

In any case, more interesting than that are a few `shufflevector` ops, some early on, that could cause problems. They are splats and broadcasts that should be possible to lower by using broadcast and insert ops.

And the key is going to be this:
```llvm
  %70 = zext <32 x i8> %56 to <32 x i32>
  %71 = sext <32 x i8> %69 to <32 x i32>
  %72 = mul <32 x i32> %70, %71
  %73 = add <32 x i32> %72, %48
```
Those four instructions are effectively a `VMAC` op. The `s/zext` ops have to be selected together with at least the `mul <32 x i32>` (into a `VMUL`), otherwise `mul` is invalid, and ideally together with the `add` op into a `VMAC`, to maximize performance.

There are a few weird things going on with insertions and extractions of `<8 x i32>` (which should be fine) and `<8 x i8>` (which might be problematic). There are lowerings we can implement that would get rid of those and replace them with insert/extract on `<32 x i32>`/`<32 x i8>` (1024b/256b), but I thought it might be an interesting issue to have as a test, since this is the common way that `MLIR` lowers multi-rank vectors from `MLIR`.

jsetoain, 2025-05-08:
I've checked again the same code, and it looks like it doesn't have any issues with the G_ADD, but it's tripping on at least one of the G_SHUFFLE. The first I see is this one:

```llvm
%80 = shufflevector <32 x i32> %73, <32 x i32> %73, <8 x i32> <i32 24, i32 25, i32 26, i32 27, i32 28, i32 29, i32 30, i32 31>
```

This is the error message:
```
fatal error: error in backend: unable to legalize instruction: %148:_(<8 x s32>) = G_SHUFFLE_VECTOR %143:_(<32 x s32>), %143:_, shufflemask(24, 25, 26, 27, 28, 29, 30, 31) (in function: conv2dk1_ui8_scalar)
```

khallouh, 2025-05-09:
I tried that shuffle instruction and it is legalized without issues (see below). Could you please share an LLVM IR reproducer on latest Peano?

```
---
name: test_shuffle
body: |
  bb.0:

    ; CHECK-LABEL: name: test_shuffle
    ; CHECK: [[COPY:%[0-9]+]]:_(<32 x s32>) = COPY $cm0
    ; CHECK-NEXT: [[COPY1:%[0-9]+]]:_(<32 x s32>) = COPY $cm0
    ; CHECK-NEXT: [[UV:%[0-9]+]]:_(<16 x s32>), [[UV1:%[0-9]+]]:_(<16 x s32>) = G_UNMERGE_VALUES [[COPY]](<32 x s32>)
    ; CHECK-NEXT: [[C:%[0-9]+]]:_(s32) = G_CONSTANT i32 8
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C]](s32)
    ; CHECK-NEXT: [[C1:%[0-9]+]]:_(s32) = G_CONSTANT i32 9
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT1:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C1]](s32)
    ; CHECK-NEXT: [[C2:%[0-9]+]]:_(s32) = G_CONSTANT i32 10
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT2:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C2]](s32)
    ; CHECK-NEXT: [[C3:%[0-9]+]]:_(s32) = G_CONSTANT i32 11
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT3:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C3]](s32)
    ; CHECK-NEXT: [[C4:%[0-9]+]]:_(s32) = G_CONSTANT i32 12
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT4:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C4]](s32)
    ; CHECK-NEXT: [[C5:%[0-9]+]]:_(s32) = G_CONSTANT i32 13
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT5:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C5]](s32)
    ; CHECK-NEXT: [[C6:%[0-9]+]]:_(s32) = G_CONSTANT i32 14
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT6:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C6]](s32)
    ; CHECK-NEXT: [[C7:%[0-9]+]]:_(s32) = G_CONSTANT i32 15
    ; CHECK-NEXT: [[AIE_SEXT_EXTRACT_VECTOR_ELT7:%[0-9]+]]:_(s32) = G_AIE_SEXT_EXTRACT_VECTOR_ELT [[UV1]](<16 x s32>), [[C7]](s32)
    ; CHECK-NEXT: [[DEF:%[0-9]+]]:_(<16 x s32>) = G_IMPLICIT_DEF
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[DEF]], [[AIE_SEXT_EXTRACT_VECTOR_ELT]](s32)
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI1:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[AIE_ADD_VECTOR_ELT_HI]], [[AIE_SEXT_EXTRACT_VECTOR_ELT1]](s32)
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI2:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[AIE_ADD_VECTOR_ELT_HI1]], [[AIE_SEXT_EXTRACT_VECTOR_ELT2]](s32)
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI3:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[AIE_ADD_VECTOR_ELT_HI2]], [[AIE_SEXT_EXTRACT_VECTOR_ELT3]](s32)
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI4:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[AIE_ADD_VECTOR_ELT_HI3]], [[AIE_SEXT_EXTRACT_VECTOR_ELT4]](s32)
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI5:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[AIE_ADD_VECTOR_ELT_HI4]], [[AIE_SEXT_EXTRACT_VECTOR_ELT5]](s32)
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI6:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[AIE_ADD_VECTOR_ELT_HI5]], [[AIE_SEXT_EXTRACT_VECTOR_ELT6]](s32)
    ; CHECK-NEXT: [[AIE_ADD_VECTOR_ELT_HI7:%[0-9]+]]:_(<16 x s32>) = G_AIE_ADD_VECTOR_ELT_HI [[AIE_ADD_VECTOR_ELT_HI6]], [[AIE_SEXT_EXTRACT_VECTOR_ELT7]](s32)
    ; CHECK-NEXT: [[UV2:%[0-9]+]]:_(<8 x s32>), [[UV3:%[0-9]+]]:_(<8 x s32>) = G_UNMERGE_VALUES [[AIE_ADD_VECTOR_ELT_HI7]](<16 x s32>)
    ; CHECK-NEXT: PseudoRET implicit $lr, implicit [[UV3]](<8 x s32>)
    %0:_(<32 x s32>) = COPY $cm0
    %1:_(<32 x s32>) = COPY $cm0
    %2:_(<8 x s32>) = G_SHUFFLE_VECTOR %0:_(<32 x s32>), %1:_, shufflemask(24, 25, 26, 27, 28, 29, 30, 31)
    PseudoRET implicit $lr, implicit %2
...

```

khallouh, 2025-05-09:
and from the IR shuffle above, We generate this on aie2p:
```
define <8 x i32> @test_shuffle(<32 x i32> noundef %73) {
; CHECK-LABEL: test_shuffle:
; CHECK:         .p2align 4
; CHECK-NEXT:  // %bb.0: // %entry
; CHECK-NEXT:    ret lr
; CHECK-NEXT:    nop // Delay Slot 5
; CHECK-NEXT:    nop // Delay Slot 4
; CHECK-NEXT:    nop // Delay Slot 3
; CHECK-NEXT:    vmov wl0, wh5 // Delay Slot 2
; CHECK-NEXT:    nop // Delay Slot 1
entry:
  %shuffle = shufflevector <32 x i32> %73, <32 x i32> %73, <8 x i32> <i32 24, i32 25, i32 26, i32 27, i32 28, i32 29, i32 30, i32 31>
  ret <8 x i32> %shuffle
}

```
and on aie2:

```
define <8 x i32> @test_shuffle(<32 x i32> noundef %73) {
; CHECK-LABEL: test_shuffle:
; CHECK:         .p2align 4
; CHECK-NEXT:  // %bb.0: // %entry
; CHECK-NEXT:    nopb ; mova r16, #8; nops ; nopx ; mov r8, r16; nopv
; CHECK-NEXT:    mova r16, #9; nopx ; vextract.s32 r0, x5, r16
; CHECK-NEXT:    mova r16, #10; vextract.s32 r1, x5, r16
; CHECK-NEXT:    vextract.s32 r2, x5, r16
; CHECK-NEXT:    mova r16, #11; vpush.hi.32 x0, x0, r0
; CHECK-NEXT:    vextract.s32 r3, x5, r16
; CHECK-NEXT:    mova r16, #12; vpush.hi.32 x0, x0, r1
; CHECK-NEXT:    vextract.s32 r4, x5, r16
; CHECK-NEXT:    mova r16, #13; vpush.hi.32 x0, x0, r2
; CHECK-NEXT:    vextract.s32 r5, x5, r16
; CHECK-NEXT:    mova r16, #14; vpush.hi.32 x0, x0, r3
; CHECK-NEXT:    vextract.s32 r6, x5, r16
; CHECK-NEXT:    mova r16, #15; vpush.hi.32 x0, x0, r4
; CHECK-NEXT:    vextract.s32 r7, x5, r16
; CHECK-NEXT:    ret lr ; vpush.hi.32 x0, x0, r5
; CHECK-NEXT:    vpush.hi.32 x0, x0, r6 // Delay Slot 5
; CHECK-NEXT:    vpush.hi.32 x0, x0, r7 // Delay Slot 4
; CHECK-NEXT:    nop // Delay Slot 3
; CHECK-NEXT:    vmov wl0, wh0 // Delay Slot 2
; CHECK-NEXT:    mov r16, r8 // Delay Slot 1
entry:
  %shuffle = shufflevector <32 x i32> %73, <32 x i32> %73, <8 x i32> <i32 24, i32 25, i32 26, i32 27, i32 28, i32 29, i32 30, i32 31>
  ret <8 x i32> %shuffle
}

```

martien-de-jong, 2025-05-09:
@jsetoain, that commit is over a year old.  I'm going to close this ticket, feel free to open a new one if you find further problemns with a recent version.

jsetoain, 2025-05-09:
@martien-de-jong What's the correct branch, then? I thought I was working with the right codebase.

jsetoain, 2025-05-09:
@martien-de-jong @khallouh 

I'm now running on `e8808be63afb2d41d98d9a6197d881acad903acb` (today's commit), and I'm seeing errors for both `aie2` and `aie2p`.

With `--target=aie2`:
```sh
fatal error: error in backend: unable to legalize instruction: %143:_(<32 x s32>) = G_ADD %142:_, %112:_ (in function: conv2dk1_ui8_scalar)
```

With `--target=aie2p`:
```sh
fatal error: error in backend: unable to legalize instruction: %142:_(<32 x s32>) = G_MUL %140:_, %141:_ (in function: conv2dk1_ui8_scalar)
```

I'll re-open the issue.