Xilinx/llvm-aie #218: Is there a tutorial about aie environment configuration?

lxy-littlechai, 2024-10-22:
Hi, I'm new to this project, and I have some troubles when compiling aie.cpp. Could you kindly tell me a tutorial or how to configure the relevant environment about aie especially the header and libraries path?

konstantinschwarz, 2024-10-25:
Hi and welcome!
Which aie.cpp are you referring to exactly? Are you trying to build LLVM? You need to configure the build using cmake, and then use ninja/make to build. Some minimal cmake configuration could look like:

```
mkdir build;
cd build;
cmake -G Ninja -C ../clang/cmake/caches/Peano-AIE.cmake ../llvm
```

You can then build clang using `ninja clang`.
I hope this helps!

flxinxout, 2025-03-17:
Hello,
I'm also getting troubles to compile a simple cpp file with the AIE API headers. My cpp file is simply:
```
#include <iostream>
#include <cstring>
#include <aie_api/aie.hpp>

using namespace aie;

int main() {

    if(arch::is(arch::AIE)) {
        std::cout << "Running on AIE\n";
    } else if(arch::is(arch::AIE_ML)) {
        std::cout << "Running on AIE-ML\n";
    }

   return 0;
}
```
The error Im getting is: `error: ‘arch’ has not been declared`

For the environement: Im using a Ryzen AI 7 with XDNA 1 (AIE-ML) NPU architecture on an Ubuntu 24.04 with a fresh XRT installation that perfectly works.  

my cmakelist file is
```
cmake_minimum_required(VERSION 3.15...3.31)

project(AMD_NPU VERSION 1.0
                DESCRIPTION "C++ Access to NPU using XRT with XDNA drivers"
                LANGUAGES C CXX)

# Required Projects
find_package(XRT REQUIRED)

# Find XRT and AIE API
set(XRT_INCLUDE_DIR "/opt/xilinx/xrt/include")
set(XRT_LIB_DIR "/opt/xilinx/xrt/lib")

# Test
add_executable(test src/test.cpp)

# Include XRT headers
target_include_directories(test PRIVATE ${XRT_INCLUDE_DIR} ${CMAKE_SOURCE_DIR}/include)

# Link against XRT libraries$
target_link_directories(test PRIVATE ${XRT_LIB_DIR})
target_link_libraries(test PRIVATE xrt_coreutil xrt++)
```

My general problem is that I don't get if I should use Vitis or not with this API. My toolchain (MLIE-AIE) says no, but it seems that the __AIE_ARCH__ cannot be defined without AIE tools. 

Thanks.