Xilinx/llvm-aie #315: Hang calling llc with aie2p target, after opt with -On n>0.

newling, 2025-01-29:
I'm starting to integrate support for aie2p into the IREE compiler. My first attempt for a small matmul is hitting a hang during linking, but only when opt is run with -On for n>0. 

I am using peano wheel from today, 28 January:  llvm_aie-19.0.0.2025012801+24e4e160.dist-info from https://github.com/Xilinx/llvm-aie/releases

input.ll is attached, and below. 

input.opt.ll with -O0 is generated with the following command:

llvm-aie/bin/opt -vectorize-loops=false -vectorize-slp=false --two-entry-phi-node-folding-threshold=10 -mandatory-inlining-before-opt=false -basic-aa-full-phi-analysis=true -basic-aa-max-lookup-search-depth=10 -O0 --inline-threshold=10 --disable-builtin=memset -S input.ll -o input.opt.ll

input.opt.ll with -O1 is generated with the following command (identical to the above, but with -O1):

llvm-aie/bin/opt -vectorize-loops=false -vectorize-slp=false --two-entry-phi-node-folding-threshold=10 -mandatory-inlining-before-opt=false -basic-aa-full-phi-analysis=true -basic-aa-max-lookup-search-depth=10 -O1 --inline-threshold=10 --disable-builtin=memset -S input.ll -o input.opt.ll

Linking works fine with the following command when run on the output of opt with '-O0':

llvm-aie/bin/llc input.opt.ll -O2 --march=aie2p --function-sections --filetype=obj -o input.o

but the above fails when run on the output of opt with '-O1':

[repro_files.zip](https://github.com/user-attachments/files/18580988/repro_files.zip)

What is causing the hang? 

input.ll:
```

; ModuleID = 'LLVMDialectModule'
source_filename = "LLVMDialectModule"
target triple = "aie2p"

@buff_11 = external global [1024 x i8]
@buff_10 = external global [1024 x i8]
@buff_9 = external global [1024 x i8]
@buff_8 = external global [1024 x i8]
@buff_7 = external global [1024 x i32]
@buff_6 = external global [1024 x i32]
@buff_5 = external global [1024 x i8]
@buff_4 = external global [1024 x i8]
@buff_3 = external global [1024 x i8]
@buff_2 = external global [1024 x i8]
@buff_1 = external global [1024 x i32]
@buff_0 = external global [1024 x i32]
@shim_2 = external global [64 x [64 x i32]]
@shim_1 = external global [64 x [64 x i8]]
@shim_0 = external global [64 x [64 x i8]]

define void @generic_matmul_0_outlined(ptr %0, ptr %1, ptr %2) {
  br label %4

4:                                                ; preds = %36, %3
  %5 = phi i64 [ %37, %36 ], [ 0, %3 ]
  %6 = icmp slt i64 %5, 4
  br i1 %6, label %7, label %38

7:                                                ; preds = %34, %4
  %8 = phi i64 [ %35, %34 ], [ 0, %4 ]
  %9 = icmp slt i64 %8, 4
  br i1 %9, label %10, label %36

10:                                               ; preds = %7
  %11 = mul nsw i64 %8, 256
  %12 = mul nsw i64 %5, 64
  %13 = add i64 %11, %12
  br label %14

14:                                               ; preds = %17, %10
  %15 = phi i64 [ %33, %17 ], [ 0, %10 ]
  %16 = icmp slt i64 %15, 4
  br i1 %16, label %17, label %34
[... 208 lines cut ...]
  call void @llvm.aie2p.release(i32 50, i32 1)
  call void @llvm.aie2p.release(i32 53, i32 1)
  %89 = add i64 %2, 2
  br label %1

90:                                               ; preds = %1
  ret void
}

declare void @llvm.aie2p.acquire(i32, i32)

declare void @llvm.aie2p.release(i32, i32)

declare <32 x i64> @llvm.aie2p.I512.I512.ACC2048.mac.conf(<16 x i32>, <32 x i16>, <32 x i64>, i32)

; Function Attrs: nocallback nofree nosync nounwind willreturn memory(inaccessiblemem: write)
declare void @llvm.assume(i1 noundef) #0

attributes #0 = { nocallback nofree nosync nounwind willreturn memory(inaccessiblemem: write) }

!llvm.module.flags = !{!0}

!0 = !{i32 2, !"Debug Info Version", i32 3}

```

martien-de-jong, 2025-01-29:
First an a-priory clarification, llc doesn't link, it transforms the llvm code to an object file.  Anyway, I'm looking at this.

martien-de-jong, 2025-01-30:
BTW, thanks for the reproducer, it hangs beautifully. FYI, it usually suffices to have the input to llc, i.e. the output of opt.

martien-de-jong, 2025-01-30:
We seem to be looping in the legalizer around G_UNMERGE_VALUES.

martien-de-jong, 2025-01-30:
The G_LOAD is a scalar load that doesn't occur in the input program. I guess it derives from scalarizing a vector load.

martien-de-jong, 2025-01-30:
OMG. We have scalarized every vector load and end up with one basic block of ~30000 instructions. We're stuck on the very first scalar load in that block. Perhaps it isn't an infinite loop, just a bit quadratic in block size.
Anyway, I think the code wasn't intended to be scalarized like this.

Well, not all. We are missing alignment here:

```
define void @generic_matmul_0_outlined(ptr nocapture readonly %0, ptr nocapture readonly %1, ptr nocapture %2) local_unnamed_addr #0 {
  .preheader:
    %.promoted = load <32 x i64>, ptr %2, align 4
    %3 = load <16 x i32>, ptr %0, align 64
    %4 = load <32 x i16>, ptr %1, align 64
    %5 = tail call <32 x i64> @llvm.aie2p.I512.I512.ACC2048.mac.conf(<16 x i32> %3, <32 x i16> %4, <32 x i64> %.promoted, i32 776)
```

%3 and %4 are properly aligned, and result in a proper vector load.
%.promoted is not aligned and gets composed from 32 scalar loads

newling, 2025-01-30:
@martien-de-jong thanks for investigating this. 

If this is an load/store alignment problem, it's not the first time for us, it's usually what trips me up lowering through peano/AIE. I've tried changing 

(1) all loads in input.ll to `align 64`
(2) all loads and stores in input.ll to `align 64`
(3)  all loads and stores and '@llvm.assume' calls in input.ll to have `align 64`

and then running with `opt` at `-O1` and then `llc`. In all cases I get an invalid instruction error in `llc` , example: `Cannot get Load Store opcode for: G_STORE %364:accregbank(<32 x s64>), %344:ptrregbank(p0) :: (store (<32 x s64>) into %ir.301, align 64)`

Can you please remind me what the alignment expectation is (or point me to docs to shed some light)?  If I can get _any_ input.ll to get through `opt` with -O1 and `llc`, then I will know what the MLIR passes we run should be generating. I'm 90% sure the code is fully vectorizable, and so the infinite/quadratic loop we're seeing is down path we shouldn't be on anyway

martien-de-jong, 2025-01-30:
I'll dig a bit further. I'm mostly looking at the llvm input to llc, which is translated to machine IR by the standard irtranslator. That initial machine IR has the 4 byte aligned load. I hope opt is cooperating in propagating alignments from input to output loads, and I hope irtranslator is doing the right thing, because that is a bit of a black box for us.
As for missing coverage of G_STORE on 32 x s64, I've been informed that this recent pr https://github.com/Xilinx/llvm-aie/pull/307 might have fixed that.

martien-de-jong, 2025-01-30:
On aie2p any vector load / store of size 512 bits or larger should have at least 64 byte alignment. In you input.ll there are several instances of 4 byte aligned loads and stores, like 
`  %.promoted = load <32 x i64>, ptr %2, align 4`

newling, 2025-01-30:
? As for missing coverage of G_STORE on 32 x s64, I've been informed that this recent pr https://github.com/Xilinx/llvm-aie/pull/307 might have fixed that.

Indeed. You folks are moving fast, it's not often that bumping a component forward by 2 days fixes anything :) I have a path forward now: fix up the alignments that our compiler is generating. And for me the 'hang' isn't high priority anymore (although of course an error for unsupported scalar code would be nice).