Xilinx/llvm-aie #352: Objdump assertion failure

jtuyls, 2025-02-13:
I am seeing the following assertion error with `llvm-objdump` on the attached object file:

[mm_npu4.o.txt](https://github.com/user-attachments/files/18782152/mm_npu4.o.txt)

```
mm_npu4.o:      file format elf32-aie

Disassembly of section .text:

00000000 <matmul_i8_i8_i32_32x32x32_8x8x8>:
       0: e1 00 00 10 00 84 00 00 00 5b 01 20 00 f0 2c 00       nopa    ;               nopb    ;               nops    ;               movxm   m1, #0x800;             nopv
      10: 7e 60 2b 00 0e 00 00 40 00 20 00 00 43 00     mova    r3, #0x2;               nopb    ;               paddxm   [sp], #0x40;           nops
      1e: 98 3d a0 10   lshl     r16, r2, r3
      22: f8 20 08 18   mov     m0, r16
      26: d4 41 00 f0 0c 41     padda    [p2], m0;              mov     m0, r0
      2c: ba 72 50 00 80 1d c7 f7 0c 01 padda    [p0], m0;              st       p6, [sp, #-60];                mov     m0, r1
      36: ba 78 60 8a ca 40 00 00 05 08 mova    r5, #0x40;              movx    r4, #0x6;               mov     r20, p2
      40: ba 78 60 48 0a 60 08 00 07 61 mova    r7, #0x308;             movx    r6, #0x100;             mov     r18, p0
      4a: 76 10 00 02 00 00 80 9d c3 f7 0c 21   padda    [p1], m0;              st       p7, [sp, #-64];                movxm   m0, #0x400
      56: ba 78 60 69 0a 21 00 00 01 00 mova    r1, #0x0;               movx    r2, #0x8;               mov     r19, p1

00000060 <.LBB0_1>:
      60: 02 70 90 34 00 60 2b 00       nops    ;               mov     p0, r18
      68: f8 20 6a 1c   mov     p4, r20
      6c: f8 20 6a 1a   mov     p2, r20
      70: e4 41 21 b0 49 08     lshl     r1, r1, r4;            mov     r0, r1
      76: e4 41 d2 b2 0a 0c     or       r16, r1, r5;           mov     p1, r18
      7c: e4 41 01 b1 47 80     lshl     r1, r16, r3;           mov     dj0, r1
      82: ba 78 10 c4 6c 01 01 70 03 00 vlda     x0, [p0, dj0];         lshl     r16, r0, r2;           mov     dj1, r16
      8c: d4 41 10 74 13 04     vlda     x2, [p0, dj1];         mov     m2, r16
      92: d4 41 01 f4 0c 49     padda    [p2], m2;              mov     m2, r1
      98: d4 01 02 f4 0c 89     padda    [p4], m2;              mov     m2, dj0
      9e: d4 01 06 f4 0c 29     padda    [p1], m2;              mov     m2, dj1
      a4: d4 81 a5 fa 0c 09     padda    [p0], m2;              mov     r21, p1
      aa: d4 81 21 0b 10 00     mova    r16, #0x0;              mov     r22, p0

000000b0 <.LBB0_2>:
      b0: 2c 00 00 b0 a2 40     vlda     bmll2, [p2, #0x0];             nopx
      b6: 98 35 15 02   vlda     bmlh2, [p2, #0x40]
      ba: 98 55 25 02   vlda     bmhl2, [p2, #0x80]
      be: 98 75 35 02   vlda     bmhh2, [p2, #0xc0]
      c2: 98 15 06 04   vlda     bmll4, [p4, #0x0]
      c6: 98 35 16 04   vlda     bmlh4, [p4, #0x40]
      ca: 98 55 26 04   vlda     bmhl4, [p4, #0x80]
      ce: d4 81 d1 be ce 86     vlda     bmhh4, [p4, #0xc0];            mov     p7, p4
      d4: d4 41 b0 f0 0c e1     padda    [p7], m0;              mov     r1, r16
      da: d4 81 c9 b6 82 e0     vlda     bmll0, [p7, #0x0];             mov     p3, p2
      e0: b6 78 d0 b4 6e 01 21 20 17 b6 86 e2   vlda     bmlh0, [p7, #0x40];            paddb    [p3], m0;              lshl     r16, r16, r2;          mov     p5, r19
      ec: ba 78 10 44 2c 13 21 b0 92 60 vlda     bmll1, [p3, #0x0];             or       r17, r16, r6;          mov     dj0, r16
      f6: d4 41 11 73 33 a0     vlda     x6, [p5, dj0];         mov     dj1, r17
      fc: 98 1b 21 05   vlda     x4, [p5, dj1]
     100: 98 b5 14 03   vlda     bmlh1, [p3, #0x40]
     104: 98 d5 24 03   vlda     bmhl1, [p3, #0x80]
     108: 98 f5 34 03   vlda     bmhh1, [p3, #0xc0]
     10c: 98 55 24 07   vlda     bmhl0, [p7, #0x80]
     110: 98 75 34 07   vlda     bmhh0, [p7, #0xc0]
     114: 08 c1 40 3b   vmac    dm3, dm2, x0, x6,r7
     118: 08 81 20 3a   vmac    dm2, dm1, x0, x4,r7
     11c: 08 c1 84 39   vmac    dm1, dm4, x2, x6,r7
     120: 22 81 04 38 e6 a0 6a 06       mov     p6, r21;                vmac    dm0, dm0, x2, x4,r7
     128: f8 a0 69 19   mov     p1, r19
     12c: f8 20 08 1a   mov     m2, r16
     130: d4 41 11 f4 0c 29     padda    [p1], m2;              mov     m2, r17
     136: ba 78 90 35 68 00 01 f0 0c a9 padda    [p5], m2;              movx    r16, #0x3;              mov     p0, r22

00000140 <.LBB0_3>:
     140: ba 00 00 00 00 00 00 70 b3 22 vlda     x6, [p1, #0x40];               nopxm
     14a: 98 1b 45 06   vlda     x4, [p6, #0x100]
     14e: 98 1b 16 05   vlda     x8, [p5, #0x40]
     152: 98 9b 46 00   vlda     x10, [p0, #0x100]
     156: 00 00         nop
     158: 18 ff 21 14   add     r16, r16, #-0x1
     15c: 84 01 40 00 00 80     jnz      r16, #0x0
     162: 00 00         nop
     164: 08 c1 68 3b   vmac    dm3, dm3, x4, x6,r7
     168: 08 01 49 3a   vmac    dm2, dm2, x4, x8,r7
     16c: 22 c1 34 39 04 f0 8c a3       padda    [p5], #0x40;           vmac    dm1, dm1, x10, x6,r7
     174: 26 01 15 38 8e 2d 8e 20 9f f0 8c c9   padda    [p6], #0x100;          vldb.4x32.lo     llvm-objdump: /project/llvm-aie/llvm/include/llvm/ADT/SmallVector.h:308: const T& llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::operator[](size_type) const [with T = llvm::MCOperand; <template-parameter-1-2> = void; const_reference = const llvm::MCOperand&; size_type = long unsigned int]: Assertion `idx < size()' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../llvm-aie/bin/llvm-objdump -d mm_npu4.o
 #0 0x000078b2a7d97f9b llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/jtuyls/workspace/llvm-aie/bin/../lib/libLLVM.so+0xb97f9b)
 #1 0x000078b2a7d9516b llvm::sys::RunSignalHandlers() (/home/jtuyls/workspace/llvm-aie/bin/../lib/libLLVM.so+0xb9516b)
 #2 0x000078b2a7d95295 SignalHandler(int) Signals.cpp:0:0
 #3 0x000078b2a6a45330 (/lib/x86_64-linux-gnu/libc.so.6+0x45330)
 #4 0x000078b2a6a9eb2c __pthread_kill_implementation ./nptl/pthread_kill.c:44:76
 #5 0x000078b2a6a9eb2c __pthread_kill_internal ./nptl/pthread_kill.c:78:10
 #6 0x000078b2a6a9eb2c pthread_kill ./nptl/pthread_kill.c:89:10
 #7 0x000078b2a6a4527e raise ./signal/../sysdeps/posix/raise.c:27:6
 #8 0x000078b2a6a288ff abort ./stdlib/abort.c:81:7
 #9 0x000078b2a6a2881b _nl_load_domain ./intl/loadmsgcat.c:1177:9
#10 0x000078b2a6a3b517 (/lib/x86_64-linux-gnu/libc.so.6+0x3b517)
#11 0x000078b2aacb4d6a llvm::AIE2PInstPrinter::printInstruction(llvm::MCInst const*, unsigned long, llvm::MCSubtargetInfo const&, llvm::raw_ostream&) (/home/jtuyls/workspace/llvm-aie/bin/../lib/libLLVM.so+0x3ab4d6a)
#12 0x000078b2aacaffa6 llvm::AIECommonInstPrinter::printInstr(llvm::MCInst const*, unsigned long, llvm::StringRef, llvm::MCSubtargetInfo const&, llvm::raw_ostream&) (/home/jtuyls/workspace/llvm-aie/bin/../lib/libLLVM.so+0x3aaffa6)
#13 0x0000000000447e04 disassembleObject(llvm::object::ObjectFile&, llvm::object::ObjectFile const&, (anonymous namespace)::DisassemblerTarget&, std::optional<(anonymous namespace)::DisassemblerTarget>&, llvm::objdump::SourcePrinter&, bool) llvm-objdump.cpp:0:0
#14 0x00000000004497a5 disassembleObject(llvm::object::ObjectFile*, bool) llvm-objdump.cpp:0:0
#15 0x000000000044c917 dumpObject(llvm::object::ObjectFile*, llvm::object::Archive const*, llvm::object::Archive::Child const*) llvm-objdump.cpp:0:0
#16 0x000000000044eca0 llvm_objdump_main(int, char**, llvm::ToolContext const&) (../llvm-aie/bin/llvm-objdump+0x44eca0)
#17 0x00000000004160a3 main (../llvm-aie/bin/llvm-objdump+0x4160a3)
#18 0x000078b2a6a2a1ca __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:74:3
#19 0x000078b2a6a2a28b call_init ./csu/../csu/libc-start.c:128:20
#20 0x000078b2a6a2a28b __libc_start_main ./csu/../csu/libc-start.c:347:5
#21 0x000000000041612e _start (../llvm-aie/bin/llvm-objdump+0x41612e)
```

martien-de-jong, 2025-02-24:
Thanks for reporting. I will look into it. I am assuming that this is aie2p, and that that architecture was selected on the command line of llvm-objdump.
I'm also assuming that the object was generated by the aie-public toolchain.

martien-de-jong, 2025-02-25:
I can reproduce on a very recent aie-public.

martien-de-jong, 2025-02-25:
Closing was inadvertent.

martien-de-jong, 2025-02-25:
There seems to be a decoder conflict between a PADDB and a VLDB_4x32. The 4x32 is one of three modes, coded in a two bit field. The fourth possibility 11 should cause the PADDB to be selected, but it doesn't.

martien-de-jong, 2025-02-26:
This is a familiar problem, where invalid values in a register operand are used to encode altogether different instructions. However, the solution used before doesn't  seem to work here.  The decoder decides on vldb.4x32.lo, but then can't decode the register values. 
Additionally, it doesn't gracefully report this (even if it shouldn't occur) but tries to print non-existing operands later on, leading to the crash.

martien-de-jong, 2025-02-26:
This is a minimal object file showing the problem. convert to ELF using yaml2obj
[x.yaml.txt](https://github.com/user-attachments/files/18984194/x.yaml.txt)

martien-de-jong, 2025-04-25:
I can fix the problem with a brute force approach: remove all cleverness about 'DecoderComplete'. We always use TryDecode, and continue down the decoder table when we get a decoder failure.
TBH, I think that the complete decoder stuff hardly saves anything in correct decodes, and isn't clever enough to flag our problem case out of the box. 
The conflict that we see is an opcode that defines some fixed bits that correspond with an illegal register in another opcode format.

martien-de-jong, 2025-04-29:
The PR mentioned above solves the issue. The offending bundle now decodes correctly to 
 `0: 26 01 15 38 8e 2d 8e 20 9f f0 8c c9   padda    [p6], #0x100;          paddb    [p0], #0x100;          padds    [p1], #0x40;           vmac    dm0, dm0, x10, x8,r7`

closed 2025-04-29 by martien-de-jong