Xilinx/llvm-aie #475: Scalar Engine `bfloat16` Issue

andrej, 2025-05-21:
There appears to be an issue with an implementation of the scalar engine handling `bfloat16`s in the kernel below.

I have encountered this with a matrix multiplication example, which works with Peano for `i8`, or with Chess with either data type (both `i8` and `bf16`).  However, using Peano and the `bf16` data type, the results are wrong.  As a side note, the bug also does not occur in Peano if the kernel uses vectorized `VMAC` instructions, so this error seems to be isolated to the scalar engine.

I have tried to simplify the breaking example to be relatively minimal (small 4x4 matmul) -- my apologies that it still seems relatively large, but a lot of it is boilerplate. I am attaching it below. Please refer to the first couple of lines of the included `Makefile`; configuring the build using `USE_CHESS=0` and `FLAGS=-DDTYPE_BF16` breaks -- other combinations work. 

[mwe.zip](https://github.com/user-attachments/files/20371675/mwe.zip)

The kernel that breaks looks like this:

```
#ifdef DTYPE_BF16
#define DTYPE bfloat16
#endif
#ifdef DTYPE_I8
#define DTYPE int8_t
#endif

// All matrices are the same square size, 4x4, for 16 elements total.
#define MATRIX_ROWS 4
#define MATRIX_COLS 4
#define SIZE MATRIX_ROWS*MATRIX_COLS

extern "C" {

void kernel_func(DTYPE *a, DTYPE *b, DTYPE *c) {
  event0();
  for (int row = 0; row < MATRIX_ROWS; row++) {
    for (int col = 0; col < MATRIX_COLS; col++) {
      DTYPE running_sum = 0;
      for (int i = 0; i < MATRIX_ROWS; i++) {
        running_sum += a[row * MATRIX_COLS + i] * b[i * MATRIX_COLS + col];
      }
      c[row * MATRIX_COLS + col] = running_sum;
    }
  }
  event1();
}

}
```

andrej, 2025-06-03:
Sorry for the delay, I will follow up with the exact outputs I'm seeing tomorrow.

konstantinschwarz, 2025-06-09:
Hi @andrej,

I think this is an issue with using functional calls within hardware loops. Could you try passing `-mllvm -enable-aie-hardware-loops=false` to see if this resolves the wrong output?

konstantinschwarz, 2025-07-07:
@andrej with https://github.com/Xilinx/llvm-aie/pull/511 being merged, you should be able to remove the `-mllvm -enable-aie-hardware-loops=false` workaround.
Is there anything left to be done here, or can this ticket be closed?