Xilinx/llvm-aie #480: `error in backend: unable to legalize instruction`

andrej, 2025-06-05:
Compiling the following minimal example with `-O2` breaks with the below attached stack trace. Annotating function `bar` with `__attribute__((noinline))` or compiling with `-O1` circumvents the issue.

```
$ cat mwe.cc 
#include <aie_api/aie.hpp>
#include <cstdint>

void bar(int8_t *in_ptr, int8_t *out_ptr) {
  aie::vector<int8_t, 16> zeroes = aie::zeros<int8_t, 16>();
  aie::store_v(out_ptr, zeroes);
}

void baz(int8_t *in_ptr, int8_t *out_ptr) {
  for (unsigned i = 0; i < 3; i++) {
    for (unsigned j = 0; j < 2; j++) {
      for (unsigned k = 0; k < 1; k++) {
        aie::vector<int8_t, 16> result = aie::add(aie::load_v<16>(in_ptr), aie::zeros<int8_t, 16>());
        aie::store_v(out_ptr, result);
      }
    }
  }
}

void foo(int8_t *in_ptr, int8_t *out_ptr) {
  bar(out_ptr, in_ptr);
  baz(in_ptr, out_ptr);
}
```

Error/backtrace:
```
$ ${PEANO_INSTALL_DIR}/bin/clang++ -O2 -std=c++20 --target=aie2-none-unknown-elf -Wno-parentheses -Wno-attributes -Wno-macro-redefined -Wno-empty-body -DNDEBUG -I /scratch/roesti/mlir-aie/install/include -c mwe.cc
fatal error: error in backend: unable to legalize instruction: %14:_(<64 x s8>) = G_SHL %4:_, %12:_(<64 x s8>) (in function: _Z3fooPaS_)
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.      Program arguments: /scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/clang++ -O2 -std=c++20 --target=aie2-none-unknown-elf -Wno-parentheses -Wno-attributes -Wno-macro-redefined -Wno-empty-body -DNDEBUG -I /scratch/roesti/mlir-aie/install/include -c mwe.cc
1.      <eof> parser at end of file
2.      Code generation
3.      Running pass 'Function Pass Manager' on module 'mwe.cc'.
4.      Running pass 'Legalizer' on function '@_Z3fooPaS_'
 #0 0x00007f7a69d99a3b llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xb99a3b)
 #1 0x00007f7a69d96f44 llvm::sys::CleanupOnSignal(unsigned long) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xb96f44)
 #2 0x00007f7a69c8ae56 llvm::CrashRecoveryContext::HandleExit(int) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xa8ae56)
 #3 0x00007f7a69d8fcde llvm::sys::Process::Exit(int, bool) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xb8fcde)
 #4 0x0000000000410225 LLVMErrorHandler(void*, char const*, bool) cc1_main.cpp:0:0
 #5 0x00007f7a69ca4df3 llvm::report_fatal_error(llvm::Twine const&, bool) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xaa4df3)
 #6 0x00007f7a6ad92e7b llvm::reportGISelFailure(llvm::MachineFunction&, llvm::TargetPassConfig const&, llvm::MachineOptimizationRemarkEmitter&, char const*, llvm::StringRef, llvm::MachineInstr const&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0x1b92e7b)
 #7 0x00007f7a6ad1394f llvm::Legalizer::runOnMachineFunction(llvm::MachineFunction&) (.part.0) Legalizer.cpp:0:0
 #8 0x00007f7a6a38884a llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (.part.0) MachineFunctionPass.cpp:0:0
 #9 0x00007f7a69f8ae00 llvm::FPPassManager::runOnFunction(llvm::Function&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xd8ae00)
#10 0x00007f7a69f8b2d1 llvm::FPPassManager::runOnModule(llvm::Module&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xd8b2d1)
#11 0x00007f7a69f8bc02 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xd8bc02)
#12 0x00007f7a7166a0e0 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::HeaderSearchOptions const&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::IntrusiveRefCntPtr<llvm::vfs::FileSystem>, std::unique_ptr<llvm::raw_pwrite_stream, std::default_delete<llvm::raw_pwrite_stream>>, clang::BackendConsumer*) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x226a0e0)
#13 0x00007f7a71b43507 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x2743507)
#14 0x00007f7a70024ebc clang::ParseAST(clang::Sema&, bool, bool) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0xc24ebc)
#15 0x00007f7a726e2a99 clang::FrontendAction::Execute() (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x32e2a99)
#16 0x00007f7a7266337e clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x326337e)
#17 0x00007f7a7277b4d8 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x337b4d8)
#18 0x0000000000411f9a cc1_main(llvm::ArrayRef<char const*>, char const*, void*) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/clang+++0x411f9a)
#19 0x000000000040b8ef ExecuteCC1Tool(llvm::SmallVectorImpl<char const*>&, llvm::ToolContext const&) driver.cpp:0:0
#20 0x00007f7a722147d9 void llvm::function_ref<void ()>::callback_fn<clang::driver::CC1Command::Execute(llvm::ArrayRef<std::optional<llvm::StringRef>>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>*, bool*) const::'lambda'()>(long) Job.cpp:0:0
#21 0x00007f7a69c8ad92 llvm::CrashRecoveryContext::RunSafely(llvm::function_ref<void ()>) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libLLVM.so+0xa8ad92)
#22 0x00007f7a72216bae clang::driver::CC1Command::Execute(llvm::ArrayRef<std::optional<llvm::StringRef>>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>*, bool*) const (.part.0) Job.cpp:0:0
#23 0x00007f7a721d52fb clang::driver::Compilation::ExecuteCommand(clang::driver::Command const&, clang::driver::Command const*&, bool) const (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x2dd52fb)
#24 0x00007f7a721d5d5e clang::driver::Compilation::ExecuteJobs(clang::driver::JobList const&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&, bool) const (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x2dd5d5e)
#25 0x00007f7a721e3664 clang::driver::Driver::ExecuteCompilation(clang::driver::Compilation&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*>>&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/../lib/libclang-cpp.so+0x2de3664)
#26 0x000000000040f940 clang_main(int, char**, llvm::ToolContext const&) (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/clang+++0x40f940)
#27 0x000000000040aef3 main (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/clang+++0x40aef3)
#28 0x00007f7a68a29d90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#29 0x00007f7a68a29e40 call_init ./csu/../csu/libc-start.c:128:20
#30 0x00007f7a68a29e40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#31 0x000000000040af3e _start (/scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin/clang+++0x40af3e)
clang++: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 19.0.0 (https://github.com/Xilinx/llvm-aie 3a817617b92fae720fb80b223e27df98c00c8c37)
Target: aie2-none-unknown-elf
Thread model: posix
InstalledDir: /scratch/roesti/buildenv/lib/python3.10/site-packages/llvm-aie/bin
Build config: +assertions
clang++: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang++: note: diagnostic msg: /tmp/mwe-fe3cec.cpp
clang++: note: diagnostic msg: /tmp/mwe-fe3cec.sh
clang++: note: diagnostic msg: 

********************
(buildenv) androsti@xcoradaie205:/scratch/roesti/mlir-aie/programming_examples/basic/combined_transpose$ 
```

martien-de-jong, 2025-06-06:
Thanks for reporting.  I'm not sure yet, but I think it might help to qualify the pointer parameters as proper vector pointers

martien-de-jong, 2025-06-06:
But perhaps I shouldn't allow you to look below the waterline... The availability of particular vector types may be target dependent, even within aie.