Xilinx/llvm-aie #487: Commit `3a81761` breaks AIE API `bf16` emulation using `bfp16` on Strix

andrej, 2025-06-12:
On Strix, for size `8x8x8`, the AIE API emulates `bf16` multiplication using `bfp16` intrinsics, see footnote (e) in the table [here](https://xilinx.github.io/aie_api/group__group__mmul.html).

Ever since [commit `3a81761`](https://github.com/Xilinx/llvm-aie/commit/3a817617b92fae720fb80b223e27df98c00c8c37), this AIE API call produces incorrect results for our matrix multiplication code when compiled using Peano on Strix. 

Prior to that commit, the code produces correct results. Happy to provide any additional information needed.

cc @L-roro

andrej, 2025-06-12:
Yes, here you go (I minimized the input source as much as I could): **[peano-issue-487.zip](https://github.com/user-attachments/files/20716597/peano-issue-487.zip)**

* `mm_chess.o` and `mm_peano_before.o` produce the correct output
* `mm_peano_after.o` produces the incorrect results
* `mm.cc` is the input source code, compiled with `${PEANO_INSTALL_DIR}/bin/clang++ -O2 -std=c++20 --target=aie2p-none-unknown-elf -Wno-parentheses -Wno-attributes -Wno-macro-redefined -Wno-empty-body -DNDEBUG -I ${MLIR_AIE_INSTALL_PATH}/include  -DAIE_API_EMULATE_BFLOAT16_MMUL_WITH_BFP16 -Dbf16_f32_ONLY -DDIM_M=32 -DDIM_K=32 -DDIM_N=32 -c mm.cc -o mm.o`

konstantinschwarz, 2025-06-12:
Thanks! How much stack do you allocate for this test?
I see that `mm_peano_before.o` uses `0x240` bytes, whereas `mm_peano_after.o` uses `0x540` bytes.

konstantinschwarz, 2025-06-12:
Great! Thanks for confirming.
Obviously this stack usage regression isn't nice.
cc @andcarminati

closed 2025-06-12 by andrej

andrej, 2025-06-12:
Excellent, that's useful info. Adding up those sizes for each function would give a lower bound. I will suggest emitting an error in our compiler if the allocated stack size is smaller than that lower bound, and potentially a warning within some threshold of it.

Thanks again for the very quick help.

andcarminati, 2025-06-13:
Hi @andrej, I see that in the "after" case, we have a more compact loop because it was pipelined, probably it leads to a faster execution too (with some stack cost). You can also disable the pipeliner using `#pragma clang loop pipeline(disable) `, but usually pipeliner does a nice job.