Xilinx/llvm-aie #540: cannot open (path_to_peano_installation)/llvm-aie/lib/clang/19/lib/aie2-unknown-none-elf/libclang_rt.builtins.a: No such file or directory

erwei-xilinx, 2025-07-14:
Since today, when running
```
(path_to_peano_installation)/llvm-aie/bin/clang -O2 --target=aie2-none-elf (path_to_design_project)/input.o -Wl,--gc-sections -Wl,--orphan-handling=error -Wl,-T,(mlir_aie_project_dir)/core_1_4.ld.script -o forward_0_core_1_4.elf
```
with nightly llvm-aie build, called from within mlir-aie, I am getting:

```
ld.lld: error: cannot open (path_to_peano_installation)/llvm-aie/lib/clang/19/lib/aie2-unknown-none-elf/libclang_rt.builtins.a: No such file or directory
ld.lld: error: unable to find library -lc
ld.lld: error: unable to find library -lm
ld.lld: error: cannot open crt0.o: No such file or directory
ld.lld: error: cannot open crt1.o: No such file or directory
clang: error: ld.lld command failed with exit code 1 (use -v to see invocation)
```

It seems that the path to `libclang_rt.builtins.a` has been changed from "aie2-unknown-none-elf" to "aie2-none-unknown-elf". May I kindly ask how we should proceed? Thanks!

konstantinschwarz, 2025-07-14:
This is related to https://github.com/llvm/llvm-project/pull/89638 which was merged into llvm-aie recently.
We typically pass a fully specified target triple, e.g. `--target=aie2p-none-unknown-elf`.

If you leave out one of the components, the default behavior to normalize target triples has changed, see also https://github.com/llvm/llvm-project/commit/16096325a5ff66d26b7e3424e7fac029c3dd35ea for the upstream announcement

erwei-xilinx, 2025-07-14:
I see, thanks for the detailed info on this, @konstantinschwarz! We will update mlir-aie to use the fully specified target triple.

closed 2025-07-14 by erwei-xilinx