Xilinx/llvm-aie #579: `"Selected slot already occupied" failed` error with today's most recent nightly build

erwei-xilinx, 2025-07-28:
Hi team,

With today's most recent nightly build, one of our designs has triggered the following error message:

```
clang++: /project/llvm-aie/llvm/lib/Target/AIE/AIEBundle.h:135: void llvm::AIE::Bundle<I>::add(I*, unsigned int, bool) [with I = llvm::MachineInstr]: Assertion `!(OccupiedSlots & NewSlots) && "Selected slot already occupied"' failed.
```

(Cmd that we ran: `{path_to_peano}/llvm-aie/bin/clang++ --target=aie2-none-unknown-elf -O2 -std=c++20 -DNDEBUG -I{path_to_mlir_aie}/mlir_aie/include -c {path_to_proj}/cascade.cc -o cascade.o`)

This error message was not triggered with last Friday's nightly build wheel. Please see code example below:

```
#define NOCPP

#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <type_traits>

#define REL_WRITE 0
#define REL_READ 1

#include <aie_api/aie.hpp>

template <typename T_in, int num_elem>
void cascade_get_impl(T_in *__restrict a) {
  event0();
  constexpr int vec_size = 16;
  T_in *__restrict pA1 = a;
  for (int i = 0; i < num_elem; i+=vec_size) {
    aie::vector<T_in, vec_size> v16 = get_scd_v16int32();
    aie::store_v(pA1, v16);
    pA1 += vec_size;
  }
  event1();
}

extern "C" {

void cascade_get(int32_t *__restrict a){
  cascade_get_impl<int32_t, 2048>(a);
}
```

We did some investigation, and noticed that the error message can be avoided if we make the following change, which makes our e2e board test to pass again:

```
template <typename T_in, int num_elem>
void cascade_get_impl(T_in *__restrict a) {
  event0();
  constexpr int vec_size = 16;
  for (int i = 0; i < num_elem; i+=vec_size) {
    aie::vector<T_in, vec_size> v16 = get_scd_v16int32();
    aie::store_v(a + i, v16);
  }
  event1();
}
```

May I ask if this error message is expected, and indicates an error in our code?

konstantinschwarz, 2025-07-28:
Hi @erwei-xilinx,
thanks for reporting this issue. We also experienced this issue in our downstream CI, and are working on a fix. Should be available tomorrow.

erwei-xilinx, 2025-07-28:
Hi @konstantinschwarz, thanks for the feedback. Looking forward to the fix.

PR #581 (2025-07-29): [AIEX] Disable the WindowPipeliner for AIE
The pass is running into an assertion in AIEBundle.h: Assertion failed: !(OccupiedSlots & NewSlots) && "Selected slot already occupied"

Disable the pass until the root cause has been fixed.

Fixes #579
closed 2025-07-29 by konstantinschwarz